UnifiedEngineering/T-962-improvements #247: Instructions for flashing seem to be wrong?

MasteringTheMess, 2023-11-20:
Having encountered the notorious "failed to autobaud" error in FlashMagic, I almost gave up until I by chance did the following:

Power off the oven
Ground n_ISP
Ground n_RESET 
Power on the oven
Release n_RESET
Release n_ISP

It was only then, that I could get FlashMagic to communicate with the LPC2134

borland1, 2023-11-22:
You didn't say what you did wrong.  "**Seem wrong**"; you must be referring to Wiki section "**Flashing the LPC21xx Controller**", heading "**Enter boot mode sequence manually**".

The instructions do work if followed, but could be improved.

To start the bootloader and enter ISP mode, it is unnecessary to power cycle the oven with the reset pin already grounded.

You can alternatively, ground the ISP pin and start the bootloader with a power cycle and skip grounding the reset pin, as turning off power essentially pulls the reset pin low.  And when power is restored, the MCU first looks for an ISP pin being low.

It is also unnecessary to "**Release ISP**" to flash the program memory.  You only need to release ISP pin from ground in order to boot the MCU normally, and test the flashed firmware.

 So, these sequences will all work to start bootloader and enter ISP mode:
1.
```
a. Power Up Oven
b. Ground ISP pin
c. Ground RESET pin momentarily
```

2.
```
a. Ground ISP pin
b. Power up oven
```

After either one of the above sequences, remove grounding of ISP pin to return to the normal boot mode. 

Only after burning the firmware in this repository, you can start the bootloader and enter ISP mode by:

3.
```
a. Press and hold the F1 button on the oven
b. Power up oven
c. Release F1 button
```

MasteringTheMess, 2023-11-23:
Yes, those are the instructions I am referring to

I followed the first sequence you describe, but FlashMagic could not connect.
The second sequence worked for me.

Since I always leave the possibility, that I did something wrong without knowing, I preferred to write that the instructions "seem" wrong...

borland1, 2023-11-24:
The number 1 sequence above is essentially the same as what you say you had success in your first post.  Did you notice by the blank oven LCD display ( or lack of a blinking red LED on the control board ), if the MCU actually booted into ISP mode?

Besides booting to ISP mode, the serial port on the PC needs proper serial driver configuration settings before the FlashMagic utility can successfully Autobaud.

MasteringTheMess, 2023-11-27:
the LED went blank after tying the ISP pin to ground in sequence 2, but I did not check the display, sorry...

serial driver must have been OK, since it worked without any further steps when using sequence 1

Maybe it would be a good idea to also describe sequence 1 as an alternative on the main article?

intreatsys, 2024-01-06:
Doooood THANK YOU.  I also had to use sequence #2 to make this work.