UnifiedEngineering/T-962-improvements #249: Heater and Fun runs togheter?

vergo77, 2023-12-14:
Hello guys, 
First of all,I would to like to thanks for your nice work!
I haved purchased the Vevor T-692 oven on Amazon (production date 22.09) and I've uploaded the latest V0.5.2 firmware, apparently with success, everything seems working properly.
But, a weird things happens and I would to like to ask if is it normal:
1 - When I switch on the oven with the internal temp greater than 45-50°C the fan starts, is it a normal behaviour?
2 - During the coocking fan and heaters runs togheter, this sounds weird for me, is it normal too?

Thanks,
Roberto

GitLang, 2023-12-14:
To accurately control the temperature of heated air, it is normal to either heat the air to excess then control it by varying the amount of fan cooling, or the other way round. The fan cools by sucking outside air into the chamber where it escapes through and round the tray and out of the bottom at the sides.

Without the cooling, it is very easy to heat something, but it will only cool at it's own, normally slow, rate. Hence the fan.

vergo77, 2023-12-14:
Many thanks for your very fast response/explanation! 
I was afraid to have damaged something! :)

Cheers