UnifiedEngineering/T-962-improvements #251: Seeking Copy of T-962a+ Firmware

GrantDraus, 2024-05-03:
Hello,

I have one of the STM32-based ovens of the T-962 family (specifically the T-962a+).

I mis-ordered the 110v variant, and expecting the 220v version, I wired it as such.

Needless to say, this fried a couple things internally, especially including the 110v-9v transformer.

I've repaired the damage, and there's only one casualty left to rectify.

The oven's firmware was erased from the MC on the control board (an STM32F103RBT6) at some point during the exposure to 220v.

The IC works perfectly well other than being erased (test firmware was successfully flashed and used).

But it of course needs to have the firmware reinstalled. 

If @xnk or anybody else has a copy of the original firmware, (or, even better- a copy of the WIP custom firmware that Mr. Johansson mentioned) 

I'd be eternally grateful.

P.S. I've also gone straight to the source & dropped Puhui themselves a line.

This was done in the interest of thoroughness, but I am not super hopeful that they'll get back to me.

Apologies if this is not the appropriate location to post this type of request, and if it isn't I'll be happy to remove it and repost somewhere better, if asked to do so. Thank you all again!

sevstels, 2024-05-15:
Somewhere in the discussion there was a file with the original firmware. Search for it.

xnk, 2024-05-15:
[t962ap.zip](https://github.com/UnifiedEngineering/T-962-improvements/files/15319454/t962ap.zip)
Not sure if it was the LPC-based FW that was previously posted, but this is the STM32-based one from back in 2021. Unfortunately the WIP firmware won't do much good yet as it only can sense temperature through an i2c-to-spi board and then the spi thermocouple interfaces (see the t962aplus branch here), but I'll hopefully get some free time to sort that out. Not sure if your EEPROM also got completely trashed during the over-voltage condition, the original eeprom dump is available [here](https://github.com/UnifiedEngineering/T-962-improvements/blob/t962aplus/STM32Cube/T962Aplus-eepromdump.txt)

borland1, 2024-05-15:
Welcome back Werner,

Like you said... in the Code tab here, there is still the unfinished **t962aplus** branch.  However that model oven only has a **three-button** switch control panel. Same button count for the **T-962 V2.0**, so that must also be **STM32-based**.   I was looking at the t962aplus code branch files earlier, and noticed that they still only supports **five-button** switch panel of earlier LPC based models.

GrantDraus, 2024-05-18:
Thank you for these files;

I'll report back with my success or failure. 

In either case I sincerely appreciate the time you took to respond! 

It's a bummer to learn that the custom firmware isn't alive yet, but I cannot complain as you all have done great work with some of the other oven variants.

P.S. @sevstels I also am deeply annoyed when people don't take the time to help themselves before begging aid.

I assure you in my case the deficit is one of competence, not effort. 

I could only find firmware in this repo for the T-962 proper, not the "-A+" variant I'm in possession of.

borland1, 2024-05-19:
You need to switch from the "Issues" tab to the T962Aplus branch in the "<> Code" tab.  You can do that by first Clicking on the "<> Code" tab near the top of this page, then select the T962Aplus branch by clicking on the dropdown tab labeled "master".

I don't think `xnk `made any promises for making a **working** T962aplus firmware version.  Because the T962aplus hardware is so different, including the limited number of control panel buttons on the T962aplus, the programming code rewrite would be too much work for very little reward.

Good luck with the repair.

TLM428, 2024-11-25:
Hey there,

Any update on the firmware for the STM32F103RBT6?