UnifiedEngineering/T-962-improvements #253: T-962 parts?

replaysMike, 2024-06-12:
Has anyone found a source for spare parts in North America?

I had just finished with flashing my own fork of the firmware, and added an external serial port and when putting it back together I ended up moving it with my hand on the open front and broke the front heating element. Broken hearted now since I spent all day upgrading it.

Alternatively, maybe I could bypass one of the elements somehow. Anyone ever tried?

GitLang, 2024-06-12:
You would be very lucky to source a spare part, but you may be able to find an element of the same rating and size. You need to make a sketch with the dimensions, and find it's power and voltage rating, and also if it is a glass bulb filament, or ceramic heater. The latter are fairly easily obtainable. If you can get a good photo or photos of an intact element then an image search on google has helped me lots of times. Actually, it's coming back to me, I think they are long cylindrical borosilicate glass tubes with the element inside. They are also sometimes called Quartz or Infra-red elements. If so, they are quite common. Whichever you use, ensure the connections are the same, or similar enough. Don't attempt the job if you are not electrically qualified and know the dangers involved. https://www.heating-elements.com/ may be able to help.  They will quote for heaters from :+1: 

Backer Hotwatt, Inc.
Danvers, MA
978-777-0070
[](https://www.hotwatt.com/)

Ulanet
Columbus, IN
812-372-0281
[](https://www.ulanet.com/)

Hi-Heat Industries, Inc.
Lewistown, MT
406-538-7411
[](https://www.hiheat.com/)

Thermo Heating Elements
Piedmont, SC
864-295-4811
[](https://thermo-llc.com/)

Dalton Electric Heating Co., Inc.
Ipswich, MA
978-356-9844
[](https://daltonelectric.com/)

Good Luck!

borland1, 2024-06-12:
$5.00 USD (plus shipping) from the oven manufacture's web site (Puhui).

[https://te168.com/products/infrared-tubes-lamps-of-puhui-reflow-oven?variant=48593619026223](url)

replaysMike, 2024-06-12:
> Actually, it's coming back to me, I think they are long cylindrical borosilicate glass tubes with the element inside. They are also sometimes called Quartz or Infra-red elements. If so, they are quite common.

Thank you that info is very helpful. They are very fragile (I barely put any pressure on it) so I do believe it's Quartz. It crumbled into small pieces, I'll have to measure the resistance to see if I can source something similar

replaysMike, 2024-06-12:
> $5.00 USD (plus shipping) from the oven manufacture's web site (Puhui).
> 

This was the first route I went, but they followed up after payment and said it would be $108USD for shipping - I could buy a new oven for that!

I did find one (literally one store) on AliExpress for $40 with shopping and ordered it, but it will likely take months to arrive.

GitLang, 2024-06-13:
Ali seem to be much quicker these days. They do a lot of stuff on their "Choice" shipping which is 5 to 7 days. They are even advertising on UK TV now!

borland1, 2024-06-13:
Yes, AliExpress.com sales have been poor, so shipping to USA by Air Mail is much faster now.

Here's an ad with price and shipping charges for one lamp to USA.  Says 110V or 220V, so they must be wired in series pairs for 220V ovens.

Note: The lamps have two nuts on each end, so tighten each end with two wrenches to avoid lamp glass breakage. 
[image]

[image]

replaysMike, 2024-07-13:
Following up - just received a new element from AliExpress today, and it's even the right one. Took 4 weeks to arrive to Canada after paying $20 shipping.

On the bright side the oven worked fine without it, I just taped off the screw contacts with kapton and it worked fine.