UnifiedEngineering/T-962-improvements #255: Is this firmware suitable for T-962 version 2.0

olegww, 2024-08-20:
Hello everyone, today I received the new version of the stove 2.0.
Has anyone tried it, is the latest firmware release suitable for this version?

<details>

<summary>Foto</summary>

[![puhui-1][1]][1]

[1]: https://github.com/user-attachments/assets/f434d823-8e77-494b-92ca-0fdea3a2b54f

[![puhui-2][2]][2]

[2]:  https://github.com/user-attachments/assets/ee669ceb-38fe-42a4-9a28-073fa3167171

[![puhui-3][3]][3]

[3]:  https://github.com/user-attachments/assets/6027b260-d910-4f03-ad20-85e2c0c97196

[![puhui-4][4]][4]

[4]:  https://github.com/user-attachments/assets/bb134778-dcb0-4d7a-82b1-c18c30f3bd5a

[![puhui-5][5]][5]

[5]:  https://github.com/user-attachments/assets/48a7a6da-e42d-435b-b68a-6e596a94a5e0

[![puhui-6][6]][6]

[6]:  https://github.com/user-attachments/assets/c0b3e4db-6a66-4031-b3ef-205fe0b3580c

</details>

GitLang, 2024-08-20:
It looks like the earth tab is STILL screwed to a painted surface, and I bet they still have earthed only one half of the case, yes?. If this is the case then please let people know and make sure your is made safe. The incoming earth/ground wire should be in good sound electrical contact with the unpainted surface of all parts of the case.

borland1, 2024-08-21:
Menus in the firmware here only supports 5-button control panels.  Your 2.0 oven only provides 3-buttons, so porting current firmware would take some effort.

There's a code branch here that was added for STM32 processors named "t962aplus", but is incomplete.

Exotelis, 2024-09-13:
I received the V2 yesterday. It looks like they've addressed some of the community's feedback. There is significantly less tape being used, but it's still present. I replaced it with kapton tape. The ground connection now at least has a washer, but the paint on the case still hasn't been removed. After I removed the paint and reconnected the ground to the case, I measured from the handle of the T-962 to ground, and everything works as it should.

I haven't tested yet whether the cold junction problem still exists - can anyone comment on that? In any case, it doesn’t really make sense to do the modifications right now without a firmware for v2, even though the pcb is still suitable for it. I'm willing to help with the firmware adjustments for V2, but only with the assistance of the maintainers (or anyone familiar with the code), so I don't have to fully dive into the project from scratch. It seems like it only affects the menu navigation. The microprocessor still appears to be the same.

- [x] Replace masking Tape
- [x] Earth connection
- [ ] Firmware v2
- [ ] Cold junction compensation

<details>
<summary>Pictures</summary>

[image]

[image]

[image]

[image]

[image]

[image]
</details>

borland1, 2024-09-13:
Exotelis,

From your photos, I don't see that unit as a T-962 2.0 oven.

The 2.0 oven has the outdated RS232 serial port, but your board photos don't show any indication such port.  Please provide a photo of your front control panel and control buttons.

Elespecia, 2024-10-03:
Have you tried the firmware or gotten any answers yet @olegww ?