UnifiedEngineering/T-962-improvements #256: Cold Junction handling

ballanty, 2024-09-11:
I've performed the CJ modification as described on the wiki page.  All is good when the T-962A is cold.  However, I can see that the CJ temperature rises during the testing and calibration runs I've been doing (as much as 10C).  It seems that the CJ and space above the over gets warmer.  I can also see that in sensor.c that the temperature estimates are continuously adjusted including a current value of the CJ temp.

I think the code assumes that the CJ temp reflects ambient temp.  I may be misunderstanding the purpose of cold junction compensation here but I don't think so.

A suggested solution is to put an action in the UI the syncs the current CJ to constant.  The op procedure then could be 'sync' the CJ temp when the oven is dead cold before running any boards.

mikeanton, 2024-09-11:
The purpose of the CJ sensor is to measure the temperature of the cold junction, not ambient.  If the cold junction changes temperature, and you don't compensate for it the actual temperature reading will be off.  The sensor you add in should be very close to where the thermocouple terminates on the PCB.  This is the location of the cold junction.  All dissimilar metal junctions on the thermocouple wire will have a voltage differential across them that varies with temperature.  Ideally this should only be where the wire is terminated, and that is what we call the cold junction, as it is on the cold end of the thermocouple.

ballanty, 2024-09-12:
Sorry, my misunderstanding of how thermocouples have to be compensated.

closed 2024-09-12 by ballanty