UnifiedEngineering/T-962-improvements #258: New motherboard? 2024 V2.0 model

the-drums, 2024-12-18:
Dear All,

Just purchased a puhui T-962 v2.0 but it appears to has a new version of motherboard.
This one also doesn't use the nxp processor but an ats amd.
Ofcourse this firmware won't work as the entire layout & processor is different. Anyone which also just purchased this model has this new style of board.

Please see the picture below.
[image]

sevstels, 2025-01-15:
Those dumb signed their own death warrant. Their oven firmware was of disgusting quality, and stoves sold and use only by upgrading the firmware. Now no one will do it for free....  And the choice of the new processor is certainly idiotic. Try to find an old board.  My condolences.

AaronJackson, 2025-02-05:
I've subscribed to this thread to see if there are any suggestions later. Out of the box though I'm quite impressed with this version 2. They made a bunch of the changes the community has suggested (No masking tape on the reflector, serial port on the side for flashing (I think), firmware seems better, and the grounding seemed good). I was quite impressed with the out of the box experience. I think it's doing some temperature compensation using the temperature at start up, but could be completely wrong. It did a good job on my first board in it though.

the-drums, 2025-02-05:
Glad to hear it was more to your liking.

Mine still had the masking tape when removing the top.
Also had a new temperature curve put on it through the pc.
The port on the side is a rs232 for software serial connection. Not flashing unfortunatly. Processor pins lead elsewhere.
The user interface is quite poor but useable. But it did reflow the board except near the end, could have placed it a tad better myself. 
Had a thermocouple on my board to check the calibration a bit an compensated the flow accordingly.
There is no option to calibrate the internal sensors of separatly for each of the tubes.
It does appear to have an onboard thermocouple aswell for junction temperature offset, though it's not on the expected location.
Regardless custom firmware would be nice, but must be rewritten for this board/processor.

djObsidian, 2025-08-08:
I haven't seen a disassembly of the new version anywhere, so I'll post it here also. 
The conclusions turned out to be premature, but in any case, the design is better than it was before.
[image]
[image]