UnifiedEngineering/T-962-improvements #259: Best configuration for 4 thermocouples

Sothis-A, 2025-06-18:
First, thanks to the community I was able to successfully mod a T962C version 2 (the one with the exhaust cuffs for vent hoses)! I made a custom PCB with 4 MAX31850K thermocouples and a USB interface onboard, and was able to flash the new firmware without much fuss.

Tonight I did my first practice run with the 'custom 2' profile and SAC305 paste. The displayed temps were WAY lower than the target curve and didn't reach 200 deg C before cooldown. However, just before the fans kicked on I heard some popping sounds inside which subsided shortly after. Upon opening the drawer I was shocked to see that solder flowed and were brown with slight charring on the pad edges. The silkscreen was yellowed, too. Clearly the oven temps were actually way too hot!! The resistors I tested with were all soldered nicely.

After letting things cool a bit I went into the bake mode menu and stumbled on what was going on. My L & R thermocouples (which are ambient like stock) were registering 50C, but my X1 and X2 thermos, which were in vias on the boards, were still measuring 100C. My assumption is that the PCB X thermos got much hotter than ambient and so the oven controller followed these temps (the code apparently follows the hottest one if its significantly higher than the others), but was displaying the L+R average (ambient) on the screen, which were lower.

This makes me wonder, what is the best configuration for accurate temps? Should I swap and make the PCB thermocouples L and R and the ambient ones X1 and X2? Or is the way I have it OK, I just need to calibrate L and R to account for them not having the PCB's thermal mass?  Or should I get rid of the ambient ones and make them all able to be attached to the PCB? Or something else even? BTW, all of the thermocouples are within 1 deg of one another measuring ambient and agree when I bring the tips together or measure ice water.

Thanks so much everyone!

GitLang, 2025-06-18:
Maybe the best way is to follow the adage of "the best part is no part". Why are you trying to use 4 thermocouples?. The 2 originals work absolutely fine, as witnessed by the many boards produced by a lot of us here and elsewhere. The Maxim chip interface seems to be a necessity though as the original thermocouple interface is poor. Can you tell us what "The resistors I tested" are?

GitLang, 2025-06-18:
The reason there are 4 is because the Maxim interface board has 4 inputs. Doesn't mean you have to use them!. I think 4 may help get a better average of the whole cabinet, but I don't think the PCB cares about individual thermocouple temperatures, just the temperature at each to-be-soldered pad. Any pad spending enough time in the right air temperature will reach that temperature, no matter if it is a feather or a block of copper. Maybe start with 2 thermocouples. Ensure it all works OK and if so add the other two in one by one. From memory, I believe the software can use the average for temperature control.

It sounds to me like your temperature feedback was reading lower than reality. That would account for the temperature display showing lower than the setup, and the actual conditions being too hot. Can you get debug information over the serial port or USB?. It would be good to log measurement of all thermocouples individually at different sets points in bake mode. If possible, an independent thermocouple and meter as a check at the same time would be useful. If that all looks OK, then from cold get set up as above and put the set point at 150C or similar and see what they read as they heat up and if they undershoot or overshoot the endpoint. If the PID is set really badly you may even see oscillation.

On a small, semi DIY prototyping oven like this I wouldn't, personally, even think of trying to measure board temperature. I have always believed this system is an air temperature control system and the board will follow on, lagging the air temperature a bit. Don't set your degC/sec slopes too high, and think air temperature. Like most temperature control of air, you have a cooling input and a heating input. You can't control without the cooling element as the natural cooling is way slower that the control loop needs to be. But it is very sensitive to PID settings. It took me a while to get it right after fitting the Maxim based board, but once right it performs great. Great, for this oven, is lagging no more than about 15C behind setpoint. It's not a problem as we're not interested in the set point curve, only the actual temperature curve. If it all happens a bit after the setpoint, what matters?. Just keep it slow, within the component soldering spec and leave a good plateau about 20 degC below liquidus. I've always got away with 60 to 90 seconds at 165 degC.

I've wandered a bit, but I'm just trying to make the point that a reliable measurement of cabinet air temperature (for this oven) is vital, and a temperature curve which reflects what is needed to do the job, is way more important than it lagging 20 degC. In an industrial soldering system you will see radiation dominant control rather than air temperature and it then become valid to monitor the actual PCB temperature. On our little oven we have to control the air and know that we will lag behind a bit, but other than a small amount of direct radiation heating, nothing on our board is going to exceed that air temperature. A *very* different situation to the IR tunnels which are almost 100% radiation dominated. I'll shut up now :)

Sothis-A, 2025-06-21:
Thank you @mikeanton & @GitLang!! It's been very tricky so far!

Just to see, I decided to swap my X1/2 (PCB) and L/R (air) TCs so the machine followed the PCB temp and the machine sort of follows the profile better but this oven has A LOT of thermal lag (my bet is the PCB's thermal lag compounds this). Despite lessening the ramp slope and upping the temps to spend more time in reflow, I now have the opposite problem in that I'm getting some good joints but other (large) areas aren't soldered. I keep upping the profile temps to spend more time in reflow. Part of this could be that I'm using the profile from my paste manufacturer (MG 4900P) but it has the profile peak briefly at 224 deg C, but apparently SAC305's liquidus temp is 220 deg C. That seems way too close, even with oven overshoot, to have enough time to reliably wet.

Have either of you tried the code modification to eliminate the PWM of the heaters? Apparently the T-962C's SSR chokes the heater current pretty bad with PWM. I found a few references for this and will likely try it.

The fans, pointing out, are pretty useless for cooling on this unit too. I'm wondering if turning one around so that it draws air in while the other exhausts isn't worth a try. Or stronger fans.

Thanks again guys, and @GitLang I appreciate the humor along with the great advice.

mikeanton, 2025-06-21:
@Sothis-A I use GC10 SAC305 solder paste, which I highly recommend if you are looking for an alternative.  It claims a liquidus temperature of 217C.   A reflow temp of 224C is closer to what you would use for leaded solder (I usually used 215C for eutectic leaded which reflowed at 183C).  Usually I shoot for a peak reflow temp of 250C for SAC305.  Generally you don't want to go over this, as many components don't like much higher than this, and I just shoot for hitting it, and not really staying there for any length of time.  Usually I'm above 217C for around a minute, which works pretty well usually, unless there are some very large components.  For large component boards, I've found it helps to have an initial preheat of around 70C for at least a minute, to get the temperature of the oven and board to not start from cold.  Then try to get the soak temp to knee over at 150C, which I often have to overdrive the air temp by about 20C or so to hit, and ramp during the soak period up to 200C before starting the reflow.  Usually I target soak from 150C to 200C to be about 90 to 120 seconds in length, longer helps to get the board to a more even temperature before you start reflow.

I don't have the control thermocouples on the PCB as that would be very difficult to manage for production, and the temperature can vary greatly across the PCB anyhow.  I usually profile the board by picking a pad on the board that is well tied to top copper, or inside planes, like a tab on a large transistor, or a power pad under a package.  Then keep adjusting the air temp profile, until the PCB temperatures match the desired profile.  You will find that different parts of the oven measure differently, so I tend to put the PCB in the same location and orientation as when I profiled the board.

I also added small diameter wires running from the front of my oven to the rear that are tensioned with springs, and these are what the board sits on.  They are raised off the bottom of the drawer by about 6mm, or 1/4".  This helps to decouple the PCB from the drawer, so that the PCB can heat up faster.  Even just putting the PCB on standoffs that are screwed into the drawer through the slots would work too.  The wires just make it easier to accommodate different PCBs without having to move the standoffs.  You can find a picture of this and further discussion about this here on another thread: [https://github.com/UnifiedEngineering/T-962-improvements/issues/228#issuecomment-1107658122](https://github.com/UnifiedEngineering/T-962-improvements/issues/228#issuecomment-1107658122).

Sothis-A, 2025-06-21:
@mikeanton That was you (the wire standoffs)!! That was a great idea. I almost stole it after seeing it but opted for some adjustable brackets instead:

[image]

As luck would have it, for bigger boards, the thumb nuts that attach the bracket to the drawer are the perfect height to support the inner PCB with that little bit of screw protruding. Too early to tell if this will work in the long run but its promising so far.

Thank you for the paste suggestion, I'll look into it! Your info also confirms I'm way too low, temp wise. Surprising that MG's recommendation is so low for the same alloy.

Sothis-A, 2025-06-21:
@GitLang OK, that likely explains the current issue of being too low temp! And ya, I do have to use LF to adhere to EU regulations. Funny, I switched to lead-free solder right at the start of my electronics journey. I still have about 1/2 a roll of Kester leaded, which was my first and only I ever bought! Despite not having any problems with the LF solder, I would love to switch back.

You both mentioned the preheat and I've noticed that, the preheat temp makes a big difference in how well the oven follows the profile. I've been trying 50-60 deg C, but I accidentally started the ramp test and the oven got to 90 deg C before I was able to stop things and load the correct profile. That nearly matched, even on the ramp up to reflow which 50 - 60 deg C really struggles to do.

sevstels, 2025-08-02:
> dual MAX6755 thermocouple board

Are you sure? MAX6755 cannot measure temperature and does not work with thermocouples

HattonLe, 2025-08-02:
> sevstels left a comment [(UnifiedEngineering/T-962-improvements#259)](https://github.com/UnifiedEngineering/T-962-improvements/issues/259#issuecomment-3146493929)

Apologies LOL... I meant 6675.

I keep getting the numbers mixed up when typing them.

Sent with [Proton Mail](https://proton.me/mail/home) secure email.