UnifiedEngineering/T-962-improvements #261: Request for Kicad schematic project files.

HattonLe, 2025-06-20:
Hi,

Is it possible you can put up the Kicad schematic files?

I have just acquired a T962 (build date looks like 09.08) and I can see a number of differences between my PCB and the T962A schematic (ignoring the heater SSR\Triac differences).

Also my PCB seems to be subtly different from photos of other PCBs as well. The main board is marked "T962/1".

I'd like to know my boards connectivity before I upload this firmware to it otherwise something might not work or need modification.

With the project files it would save me starting from scratch as I work my way through my boards layout.

Thanks.

borland1, 2025-06-20:
First identify which processor and number of control panel buttons your unit has.  Upload photos of your PCB board.

If you have NXP's LPC21xx processor and panel has five buttons, firmware here should work fine.

HattonLe, 2025-06-20:
NXP LPC2144F8064 running at 11.0592Mhz. 128k flash, 16k RAM, 2 ADCs with 14 channels overall, 1 DAC.
LCD display is a Waying WYM12684D2.

[image]
[image]

The PCB actually has a properly labelled "JTAG" connector.

Manufacturer wiring hack for powering system fan and reuse of a JTAG pin with a S8550 possibly for use as an interrupt from the S key???
Also note strange unused "RW" and "RW2" signals. Not sure what they might have been intended for.

I also note U9 pin 5 is a floating input - that should be tied to Gnd as best practice.

Component IDs also differ from the T962A schematic. I have spare time to make and upload a T962/1 specific schematic if I don't have to start doing one from scratch, hence me asking.

HattonLe, 2025-08-02:
Don't bother. I did my own from scratch.

closed 2025-08-02 by HattonLe